juanmf/StepperMotors #1: Support for Adafruit Motorkit

tomhaines, 2024-01-13:
Do you have plans to add support for the [Adafruit Motorkit](https://www.adafruit.com/product/2348?gad_source=1&gclid=CjwKCAiA44OtBhAOEiwAj4gpOX0WuayaWbzV0AFW309iVW0I8tnOyNXCGX3p1VVwOHkeFpPK8P1TwhoCqjUQAvD_BwE)?

I am using this extensively in a project, but the supplied MotorKit libraries are fairly rudimentary. I would be happy to help test on my hardware.

juanmf, 2024-01-13:
It should be fairly simple to add the stepper side of it, given:
*  Pinout description. (direction, step, microstep mode pins, sleep, halt, etc) & their on/off values (HIGH, LOW)
*  micro-stepping pins values => mode resolution

For you to test, also provide your motor's:
* model name
* minPPS & maxPPS
* SPR (steps per rev.)

  Currently I'm focusing on my main project, this is a spin-off. So I'm frequently adding features as needed from a client side perspective. For more hardware implementations I'd appreciate the details requested above so I can just write the code.
  
 Once I provide these two implementations it should be straight forward for you to test. I have only used full-step mode because I need the torque. So If you want to test micro-stepping too, that'd be cool. 
Thanks

tomhaines, 2024-01-13:
Thanks for digging into this. I use this stepper board because it uses I2C for the interface, leaving the rest of the GPIO pins for my project, which requires all but one of the pins.

juanmf, 2024-01-13:
It's a bit out of scope. 
But if you think it'd help more people, and provide the PWM details for me, I'll see to implement a H-Bridge based driver Strategy. Not trivial.

juanmf, 2024-02-23:
Implemented an Adapter for Adafruit drivers, `MotorKit.stepper1` `MotorKit.stepper2` (Experimental).
See demo code bellow, Haven had a chance to test it yet (Hardware shipping).

https://github.com/juanmf/StepperMotors/commit/b02db1bc19e24cf342dd9db624a7dde7c6636a6b

```Python
"""
Demonstrate how to build and use Adafruit drivers, wrapped with a BipolarStepperMotorDriver
and all it's acceleration strategies.
"""
import board
from adafruit_motor import stepper
from adafruit_motorkit import MotorKit
from stepper_motors_juanmf1.AdafruitMotorAdapter import AdafruitStepperAdapter
from stepper_motors_juanmf1.Controller import BipolarStepperMotorDriver
from stepper_motors_juanmf1.ControllerFactory import DynamicControllerFactory, StaticControllerFactory, \
    SynchronizedControllerFactory
from stepper_motors_juanmf1.StepperMotor import Nema17_42Ncm_17HS4401

from src.stepper_motors_juanmf1.Navigation import BasicSynchronizedNavigation

SUPPORTED_STEPPING_STYLES_OR_MODES = AdafruitStepperAdapter.ADAFRUIT_STYLES + AdafruitStepperAdapter.RESOLUTION.keys()

# Threads in same process (inefficient)
# factory = StaticControllerFactory()        # Non-interruptible stepping jobs
# factory = SynchronizedControllerFactory()  # Interruptible stepping jobs; managed in a centralized way,
                                             #   no competing stepping threads.
factory = DynamicControllerFactory()         # Interruptible stepping jobs

kit = MotorKit(i2c=board.I2C())
adafruitDriver = kit.stepper1
stepMode = SUPPORTED_STEPPING_STYLES_OR_MODES[SUPPORTED_STEPPING_STYLES_OR_MODES.index[stepper.MICROSTEP]]

motor = Nema17_42Ncm_17HS4401(loaded=True)
adafruitAdapter = factory.getExponentialAdafruitStepperWith(stepperMotor=motor,
                                                            adafruitDriver=adafruitDriver,
                                                            stepsMode=stepMode)

adafruitAdapter.signedMicroSteps(steps=-800)
```

juanmf, 2024-02-27:
Got it working wit latest commit 6e8d1e3.
As in no crashes and the motor trying to do something. But it heats a lot and probably the configured PPS was off. 
I hear the motor pulsing, in linear acceleration with an increasingly fast pace, but nowhere near the 50PPS I set for the motor.
```Python

import board
from adafruit_motorkit import MotorKit
...

    def _initLocalDrivers(self, controllerFactory: ControllerFactory):
        nema = GenericStepper(minPps=50, maxPps=80, minSleepTime=1/80, maxSleepTime=1/50)
        nema2 = GenericStepper(minPps=50, maxPps=80, minSleepTime=1/80, maxSleepTime=1/50)

        kit = MotorKit(i2c=board.I2C())
        adafruitDriver1 = kit.stepper1
        adafruitDriver2 = kit.stepper2

        self.azimuthDriver = (ControllerBuilder()
                                .getBasicBuilder(nema, directionGpioPin=None, stepGpioPin=None, stepsMode="1/16")
                                .withNavigationStyleSynchronized()
                                .withAdafruitDriver(adafruitDriver1)
                                .withLinearAcceleration()
                                .buildAdafruitStepperDriverAdapter())

        self.elevationDriver = (ControllerBuilder()
                                .getBasicBuilder(nema2, directionGpioPin=None, stepGpioPin=None, stepsMode="1/16")
                                .withNavigationStyleSynchronized()
                                .withAdafruitDriver(adafruitDriver2)
                                .withLinearAcceleration()
                                .buildAdafruitStepperDriverAdapter())
```

juanmf, 2024-03-01:
With this I consider support for Adafruit added.
[doc/demo/AdafruitStepperAllCombinationsDemo.py](https://github.com/juanmf/StepperMotors/blob/cf9aa2ef5b02a1650e45b80eb6254ae492e7a941/doc/demo/AdafruitStepperAllCombinationsDemo.py) tested All (most) possible combinations of [accelerations, navigation, allSteppingModes] with no apparent errors. 

Output:
```Python
>>> drivers
{'withLinearAcceleration-withNavigationStyleStatic-Full': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851bd850>, 'withLinearAcceleration-withNavigationStyleStatic-Half': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851be2d0>, 'withLinearAcceleration-withNavigationStyleStatic-1/8': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851bed90>, 'withLinearAcceleration-withNavigationStyleStatic-1/16': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851bf750>, 'withLinearAcceleration-withNavigationStyleStatic-1': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851bffd0>, 'withLinearAcceleration-withNavigationStyleStatic-2': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851c8c90>, 'withLinearAcceleration-withNavigationStyleStatic-3': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851bc6d0>, 'withLinearAcceleration-withNavigationStyleStatic-4': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851c9b90>, 'withLinearAcceleration-withNavigationStyleDynamic-Full': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851ca610>, 'withLinearAcceleration-withNavigationStyleDynamic-Half': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851cb090>, 'withLinearAcceleration-withNavigationStyleDynamic-1/8': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851cbb50>, 'withLinearAcceleration-withNavigationStyleDynamic-1/16': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851dc650>, 'withLinearAcceleration-withNavigationStyleDynamic-1': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851dcfd0>, 'withLinearAcceleration-withNavigationStyleDynamic-2': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851dda90>, 'withLinearAcceleration-withNavigationStyleDynamic-3': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851de510>, 'withLinearAcceleration-withNavigationStyleDynamic-4': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851defd0>, 'withLinearAcceleration-withNavigationStyleSynchronized-Full': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851e8d50>, 'withLinearAcceleration-withNavigationStyleSynchronized-Half': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851e9790>, 'withLinearAcceleration-withNavigationStyleSynchronized-1/8': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851ea210>, 'withLinearAcceleration-withNavigationStyleSynchronized-1/16': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851eac90>, 'withLinearAcceleration-withNavigationStyleSynchronized-1': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851eb710>, 'withLinearAcceleration-withNavigationStyleSynchronized-2': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851ebfd0>, 'withLinearAcceleration-withNavigationStyleSynchronized-3': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f4cd0>, 'withLinearAcceleration-withNavigationStyleSynchronized-4': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f5750>, 'withExponentialAcceleration-withNavigationStyleStatic-Full': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f6290>, 'withExponentialAcceleration-withNavigationStyleStatic-Half': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f6cd0>, 'withExponentialAcceleration-withNavigationStyleStatic-1/8': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f77d0>, 'withExponentialAcceleration-withNavigationStyleStatic-1/16': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f851f7fd0>, 'withExponentialAcceleration-withNavigationStyleStatic-1': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f85204dd0>, 'withExponentialAcceleration-withNavigationStyleStatic-2': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f852057d0>, 'withExponentialAcceleration-withNavigationStyleStatic-3': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f852062d0>, 'withExponentialAcceleration-withNavigationStyleStatic-4': <stepper_motors_juanmf1.Adafr [... cut ...]
    'withExponentialAcceleration-withNavigationStyleSynchronized-4': <stepper_motors_juanmf1.AdafruitControllerAdapter.AdafruitStepperDriverAdapter object at 0x7f85231950>
}
>>> l = None
# Motor visibly started and stopped on each round.
>>> for d in drivers.values():
...     d.signedSteps(200).result()
...     l = d
... 
695281380143
696760044376
698229875374
699708866664
701188885613
[... 20 lines cut ...]
730447405533
731546065631
732660912385
733853732568
735038030339
736143681742
737248929305
738422708835
739597037309
740773292879
741890853081
743064757993
744239750046
745414264224
746579366052
True
True
True
True
True
True
True
True
>>> l.release()
```

closed 2024-03-01 by juanmf